fluidd-core/fluidd #534: Small Feature Request

Dreistein75, 2022-02-22:


federico9292, 2022-06-16:
since it's like a graphic-only request and doesn't involve much security or checks or many validations, I think perhaps I might take a look at it...?

matmen, 2022-06-20:
@federico9292 Feel free to work on it if you want, community contributions are highly appreciated! If you need any help just let us know in our Discord server; you can find development instructions here: https://docs.fluidd.xyz/development

ChaosBlades, 2022-08-30:
> Just wondering: I assume the RPi and the Octopus are both purple, with one being slightly lighter?

I want to say it used to be but not anymore or it depends on your config. Mine currently looks like this.
[image]

thijstriemstra, 2023-02-23:
+1 for this feature! FWIW, KlipperScreen supports this feature, e.g. https://github.com/jordanruthe/KlipperScreen/blob/v0.3.1/styles/base.conf

```conf
{
    "graph_colors": {
        "extruder": {
            "colors": ["FF5252", "FF7252", "FF9252", "FFB252", "FFD252"],
            "state": 0
        },
        "bed": {
            "colors": ["1FB0FF"],
            "state": 0
        },
        "fan": {
            "colors": ["3DC25A", "58FC7C", "10EB40", "7EF297"],
            "state": 0
        },
        "sensor": {
            "colors": ["D67600", "830EE3", "B366F2", "E06573", "E38819"],
            "state": 0
        }
    }
}
```

MxBrnr, 2025-04-13:
This should really be considered an accessibility issue rather than convenience or QOL. I greatly prefer Fluidd to Mainsail, but the ability to set custom colors to the thermals graph (as well as other areas of the UI), along with being able to change the vertical size of the graph, are the two main things preventing me from switching to Fluidd; the graph is just too difficult for me to read in its current implementation without these options.