fluidd-core/fluidd #536: Update manager and restart option for klipperscreen

behold81, 2022-02-23:
### Is your feature request related to a problem? Please describe

To have the ability to update klipper screen from within fluidd along with a restart option. 

### Describe the solution you'd like

Under restart section to have a klipperscreen restart
Under updates have klipperscreen

matmen, 2022-02-24:
Fluidd already has the ability to update klipperscreen built in via moonraker: https://github.com/jordanruthe/KlipperScreen/blob/master/docs/Installation.md#klipper-and-moonraker-installation
The option to restart the klipperscreen service is coming [Fluidd v1.17](https://github.com/fluidd-core/fluidd/pull/504) (ETA end of week)

closed 2022-02-24 by matmen

behold81, 2022-02-24:
Thank You. Not sure how I missed the updater. I am looking forward to 1.17. I really do like fluidd and glad its starting to be developed again. Appriciated.