fluidd-core/fluidd #540: Export / Import history and printer stats

darkblaze69, 2022-02-25:
### Is your feature request related to a problem? Please describe

Cannot find a way to migrate my history and printer stats like total time / jobs etc during migration to a newer hardware / OS re-installation.

### Describe the solution you'd like

Needs Export / Import history and printer stats.
Needed for example in case with installation migration to a newer hardware.

matmen, 2022-02-25:
I believe all of this is stored by moonraker in `/home/pi/.moonraker_database/`, so I'm not sure if this is in scope for Fluidd. I'll talk to the team and see what they think about this.

pedrolamas, 2022-03-03:
This is not something I believe we can do from Fluidd as those are Moonraker stats.

Moonraker [History APIs](https://moonraker.readthedocs.io/en/latest/web_api/#history-apis) only allow to get and reset the values.

closed 2022-03-03 by pedrolamas