fluidd-core/fluidd #542: Fullscreen Camera View not respecting Settings

Durahl, 2022-02-26:
### What happened

Greetings!

[Due to how I mount my Camera in my Frame](https://i.imgur.com/3NKCEuY.jpg) I need _both_ Camera Axis flipped for the Camera View to match what I'd see standing in front of the machine.

The settings allow for such an adjustment and the Regular Camera View respects that setting but the Fullscreen Camera View does not.

### What did you expect to happen

I'd expect the Fullscreen Camera View to also be respecting the in the Camera Settings made changes.

### How to reproduce

Set up a Camera and in the Camera Settings have one **or** the other **or** both Camera Axis flipped:
The Regular Camera Module **will** respect those changes.
The Fullscreen Camera View **will not** respect those changes.

### Additional information

Nuffin...

matmen, 2022-02-26:
The fullscreen camera feed only opens the raw stream in a new tab at the moment, we can't change the orientation for that. We can however implement it for the "embedded" full screen view, most likely coming soon (for reference: https://github.com/fluidd-core/fluidd/issues/279#issuecomment-1046342209)

matthewlloyd, 2022-02-27:
Until then, another option to edit your mjpg_streamer command line options (assuming you're using it - the standard OctoPi, FluiddPi, MainsailOS etc. distributions all do) so that the transformation you need is applied directly at the source. The `-hf` option can apply horizontal flip, `-vf` vertical flip, and the `-rot` option rotation `0-359` degrees: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/README.md

For example if your mjpg_streamer command line looks like this:

`mjpg_streamer -i 'input_uvc.so [options]' ...`

You could change it to:

`mjpg_streamer -i 'input_uvc.so -hf -vf [options]' ...`

Durahl, 2022-02-27:
Where exactly would that command line have to be edited?
It doesn't appear to be part of the `webcam.txt` found in **FluiddOS** 🤔

matthewlloyd, 2022-02-27:
@Durahl I don't use FluiddPi but I think you can add uncomment `camera_usb_options` in `/home/pi/klipper_config/webcam.txt` and add the options there:

https://github.com/fluidd-core/FluiddPI/blob/master/src/modules/mjpgstreamer/filesystem/home/pi/klipper_config/webcam.txt

Durahl, 2022-03-02:
@matthewlloyd Yea, I was already experimenting with that line but without success...
I tried positioning the -vf -hf in various positions of the code without success - Any suggestions what I'm doing wrong?

matthewlloyd, 2022-03-20:
The -hf and -vf probably need to go inside the quotes:

camera_usb_options="-r 2560x1440 -f 60 -hf -vf"

Durahl, 2022-03-22:
@matthewlloyd Nope, doesn't work - The Fullscreen View ignores _that_ setting too =)

PR #609 (2022-06-16): feat: fullscreen camera page
Adds a dedicated page for the fullscreen camera feed, solving the following issues:
- Camera settings are now respected (rotation, fps, ...)
- The emergency stop button (as well as all other menu items) are now available while being in the full screen view

Currently still opens in a new tab, not sure if this is the appropriate behavior

Tested with (adaptive and streamed) MJPEG sources on:
- Chrome Desktop
- Chrome Mobile (both landscape and portrait orentation)
- Firefox Desktop
- Edge Desktop

[image]

Closes #542 #279
closed 2022-06-16 by matmen