fluidd-core/fluidd #545: The new feature for displaying an alias for a macro doesn't work, displayed name remains the actual macro name

mjparme, 2022-02-28:
### What happened

I have added Aliases to all my Macro's in the newest version of Fluidd. However, the displayed text remains the upper cased named from the printer.cfg file.

### What did you expect to happen

I expected the alias to be displayed rather than the actual macro named from the printer.cfg.

### How to reproduce

Goto Settings -> Macros. left-click on one of the macros. Fill in an alias for the macro in the Alias text field. Click on Home then view the Macro section and notice the Alias is not displayed

[image]
[image]
.

mjparme, 2022-02-28:
The version of fluidd in the footer is `© 2022 fluidd: v1.17.0-85bacdb`

mjparme, 2022-02-28:
I tested in a different browser that I have never opened Fluidd in before to see if it was a local cache issue. However, the problem remains.

Biorn1950, 2022-02-28:
Hello what is reported by http://fluiddpi.local:7125/server/database/item?namespace=fluidd

matmen, 2022-02-28:
I ran into the same issue when updating to v1.17.0, but IIRC I could solve it by restarting moonraker.

mjparme, 2022-02-28:
@Biorn1950 here is the output you requested

```json
{
   "result":{
      "namespace":"fluidd",
      "key":null,
      "value":{
         "cameras":{
            "cameras":[
               {
                  "id":"03674a32-77a5-49a0-b158-04d07c198bfd",
                  "enabled":true,
                  "name":"Default",
                  "type":"mjpgadaptive",
                  "fpstarget":15,
                  "url":"/webcam/?action=stream",
                  "flipX":false,
                  "flipY":false
               }
            ]
         },
         "charts":{
            "selectedLegends":{
               "heater_bed":true,
               "heater_bedTarget":true,
               "heater_bedPower":false,
               "extruder":false,
               "extruderTarget":false,
               "extruderPower":false
            }
         },
         "console":{
            "commandHistory":[
               "g1 Z0 F500",
               "g90",
               "g1 Z0 F500",
               "g90",
               "g1 Z0 F500",
               "g90",
               "save_config",
               "accept",
               "testz z=0.025",
               "testz z=0.025",
               "testz z=-0.076",
               "testz z=-0.025",
               "testz z=-0.076",
               "testz z=-0.1",
               "testz z=-0.1",
               "testz z=-0.1",
               "testz z=-0.5",
               "testz z=-1",
               "testz z=-4.978",
               "probe_calibrate"
            ]
         },
         "layout":{
            "layouts":{
               "dashboard":{
                  "container1":[
                     {
                        "id":"printer-status-card",
                        "enabled":true,
                        "collapsed":false
                     },
                     {
                        "id":"camera-card",
                        "enabled":true,
                        "collapsed":true
                     },
[... 208 lines cut ...]
                     "visible":false
                  },
                  {
                     "value":"filament_weight_total",
                     "visible":false
                  },
                  {
                     "value":"object_height",
                     "visible":true
                  },
                  {
                     "value":"first_layer_height",
                     "visible":false
                  },
                  {
                     "value":"layer_height",
                     "visible":false
                  }
               ]
            }
         }
      }
   }
}
```

mjparme, 2022-02-28:
> I ran into the same issue when updating to v1.17.0, but IIRC I could solve it by restarting moonraker.

I will give this a try.

mjparme, 2022-02-28:
I restarted Klipper and Moonraker and the problem remained, I then went ahead and rebooted the PI just so everything restarted and the problem still remains.

[image]

mjparme, 2022-02-28:
@Biorn1950 that is what I get for procrastinating on categorizing my macros :-) Thanks for the fix.

PR #548 (2022-02-28): Fix: Alias for uncategorized macro
Fix #545

Signed-of-by: Biorn1950 <biorn@me.com>
closed 2022-02-28 by matmen