fluidd-core/fluidd #561: There are typos in the assignments tab for the german version

FireN1337, 2022-03-05:
### What happened

There are typos in the german translation regarding "temperature".

"Temperature" should be "Temperatur" in german, but it says "Temparatur" for the bed and "Temparatiur" for the extruder.

:)

### What did you expect to happen

I did not expect anything out from typos :)

### How to reproduce

turn german on :)

### Additional information

It´s not a high priority bug, but maybe you have time to fix it in between some things.
Love Fluidd and appreciate the work you put in to :)

pedrolamas, 2022-03-06:
Hi @FireN1337, thank you for your comments and pointing this out.

Translations are mostly result from community Pull Requests - we don't have the manpower or even the knowledge to evaluate if all those are correct or not, so we mostly just ensure that they don't break the layout.

If you do see places where these translations can be improved, I invite you to fix and send a PR so we can review and merge. Thank you!

closed 2022-03-06 by matmen

FireN1337, 2022-03-07:
If I knew where to look for the file where I could change this, I would have done this a long time ago.
If you provide me with a path and file name to look for, I would gladly correct all language faults for the german version if I have some free time. (Where should I post this if I'm done? - Pull Requests?) 
Im new to this kind of github stuff.

greetings :)

PR #562 (2022-03-06): chore(i18n): improve, fix and finish german translations
Adds all missing translations for current features (except for `app.console.placeholder.scroll`, which doesn't seem to be used anymore).  Fixes and improves some other translations (see linked issue) (some changes only for layout fixes).

Closes #561