fluidd-core/fluidd #564: Full Config Folder Download - zip/tar for archiving

marc-gist, 2022-03-06:
### Is your feature request related to a problem? Please describe

In the event of an SSD card failure, one should have a backup of the printers config files.  A great deal of effort can go into tweaking a printers configuration. I would be nice to have an easy way to download the entire config folder.

### Describe the solution you'd like

A button that would trigger a `tar pcfz backup-<datestamp>.tgz ~/klipper_config/` and either auto download, or add the file to a folder that is accessible by Fluidd for the user to download and store outside/off the printer/pi's file system.

### Describe alternatives you've considered

Currently I ssh into the machine and do the command myself, then move the file to where I can easily download via fluidd.
works, but for others not as used to SSH/linux commands a quick and easy way would be nice for everyone to have.

pedrolamas, 2022-03-06:
Makes sense and something I do believe could be quite useful for every user!

Having said that, Fluidd is just a web interface running on the client-side browser... so, this first needs to be implemented on [Moonraker](https://github.com/Arksine/moonraker/) (the server-side API) for us to be able to expose such a feature!

You will need to open this request there and if it gets implemented there, feel free to re-open this ticket so we can prioritize and implement!

closed 2022-03-06 by pedrolamas

marc-gist, 2022-03-06:
Thank you for the quick response, but I disagree that this should be implemented in Moonraker first... although that would be neat as well to be able to use the API to backup files, its a user interface feature... although i guess if its in the API it could be automated easier.
I'll make a request on Moonraker, but hope you reconsider here.

matmen, 2022-03-06:
Fluidd has no means of talking to the host machine other than through Moonraker, so an implementation of this in Fluidd would need to iterate over all the directories and have a browser-based zip implementation, which I think would bloat the bundle size a lot for just a small feature. Moonraker generating a zip server-side would add practically no overhead for Fluidd, and it would be a lot faster and more reliable.

ChaosBlades, 2022-10-11:
Here is the moonraker issue for this for anyone else that landed here looking for this feature.
https://github.com/Arksine/moonraker/issues/259