fluidd-core/fluidd #571: DNS doesn't work correctly

jagauthier, 2022-03-12:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Moonraker is in a docker container, called "printer"
Fluidd is in another docker container called "fluidd"
When attempting to add the printer to fluidd:
[image]

Adding the printer by IP works, but is not a desired solution.
DNS works properly from the command line of the container:

```
# nslookup printer1
Server:         127.0.0.11
Address:        127.0.0.11#53

Non-authoritative answer:
Name:   printer1
Address: 172.19.0.3

```

### What did you expect to happen

Expected to be able to lookup containers by name.

### How to reproduce

Put both moonraker and fluidd in different containers within the same docker network.

matmen, 2022-03-12:
Hey, can you attach a screenshot of the browser's dev console output (showing the exact error message for the failed network request)? I believe this may be a [CORS issue](https://developer.chrome.com/blog/private-network-access-update/)

jagauthier, 2022-03-12:
[image]

I was under the impression that fluidd was doing the resolution. If it's happening at the browser level that is why.
I will have to create a resolution on my local network for it. At the moment it's just internal docker resoution.

pedrolamas, 2022-03-12:
As you've noticed now, this is not a Fluidd bug but the result of the way you are building your Docker network stack.

Docker hosts are normally not resolved by the host, so this is expected behavior!

jagauthier, 2022-03-12:
Correct. This was my lack of understanding that it was the web app resolving and not happening inside the container.  Now I understand how the fluidd network stack works better., The issue is easily resovable.

closed 2022-03-12 by jagauthier