fluidd-core/fluidd #625: Same as here: Webcam Window is Tiny #371

homuser2003, 2022-04-09:
### Browser

Edge

### Device

Laptop

### Operating System

Windows

### What happened

Both in Edge and Chrome - The webcam is appearing only collapsed on the dashboard but works fine when I click on the fullscreen button. The solution described here "Webcam Window is Tiny #371" does not seem to work for me. See config files below. Thanks!

Camera ist identified by lsusb as: Bus 001 Device 005: ID 0bda:2137 Realtek Semiconductor Corp. Web Camera

**Printer.cfg**
[include kiauh_macros.cfg]
[mcu]
serial: /dev/serial/by-id/<your-mcu-id>

[pause_resume]

[display_status]

[virtual_sdcard]
path: ~/gcode_files

[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
microsteps: 16
rotation_distance: 32
endstop_pin: ^!PC2
position_endstop: -30
position_max: 220
position_min: -30
homing_speed: 80

[stepper_y]
step_pin: PC6
dir_pin: PC7
enable_pin: !PD6
microsteps: 16
rotation_distance: 32
endstop_pin: ^!PC3
position_endstop: -8
position_min: -8
position_max: 220
homing_speed: 80

[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
position_min: -2 # The Z carriage may need to travel below the Z=0
                 # homing point if the bed has a significant tilt.
#endstop_pin: ^!PC4
#position_endstop: 0.5
position_max: 240
homing_speed: 10

[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
microsteps: 16
rotation_distance: 33.600
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA7
control: pid
pid_Kp: 2.151492
pid_Ki: 0.633897
pid_Kd: 230.042965
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA6
control: watermark
min_temp: 0
max_temp: 105

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 1000
max_z_velocity: 20
max_z_accel: 100

[display]
lcd_type: hd44780
rs_pin: PA3
e_pin: PA2
d4_pin: PD2
d5_pin: PD3
d6_pin: PC0
d7_pin: PC1
up_pin: PA1
analog_range_up_pin: 9000, 13000
down_pin: PA1
analog_range_down_pin: 800, 1300
click_pin: PA1
analog_range_click_pin: 2000, 2500
back_pin: PA1
analog_range_back_pin: 4500, 5000
#kill_pin: PA1
#analog_range_kill_pin: 400, 600

[probe]
pin: ^PC4
#   Probe detection pin. If the pin is on a different microcontroller
#   than the Z steppers then it enables "multi-mcu homing". This
#   parameter must be provided.
#deactivate_on_each_sample: True
#   This determines if Klipper should execute deactivation gcode
#   between each probe attempt when performing a multiple probe
#   sequence. The default is True.
x_offset: -22
#   The distance (in mm) between the probe and the nozzle along the
#   x-axis. The default is 0.
y_offset: -36
#   The distance (in mm) between the probe and the nozzle along the
#   y-axis. The default is 0.
#z_offset: -1
#   The distance (in mm) between the bed and the nozzle when the probe
#   triggers. This parameter must be provided.
speed: 3.0
#   Speed (in mm/s) of the Z axis when probing. The default is 5mm/s.
samples: 3 
#   The number of times to probe each point. The probed z-values will
#   be averaged. The default is to probe 1 time.
#sample_retract_dist: 2.0
#   The distance (in mm) to lift the toolhead between each sample (if
#   sampling more than once). The default is 2mm.
#lift_speed:
#   Speed (in mm/s) of the Z axis when lifting the probe between
#   samples. The default is to use the same value as the 'speed'
#   parameter.
#samples_result: average
#   The calculation method when sampling more than once - either
#   "median" or "average". The default is average.
#samples_tolerance: 0.100
#   The maximum Z distance (in mm) that a sample may differ from other
#   samples. If this tolerance is exceeded then either an error is
#   reported or the attempt is restarted (see
#   samples_tolerance_retries). The default is 0.100mm.
samples_tolerance_retries: 1
#   The number of times to retry if a sample is found that exceeds
#   samples_tolerance. On a retry, all current samples are discarded
#   and the probe attempt is restarted. If a valid set of samples are
#   not obtained in the given number of retries then an error is
#   reported. The default is zero which causes an error to be reported
#   on the first sample that exceeds samples_tolerance.
#activate_gcode:
#   A list of G-Code commands to execute prior to each probe attempt.
#   See docs/Command_Templates.md for G-Code format. This may be
#   useful if the probe needs to be activated in some way. Do not
#   issue any commands here that move the toolhead (eg, G1). The
#   default is to not run any special G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands to execute after each probe attempt
#   completes. See docs/Command_Templates.md for G-Code format. Do not
#   issue any commands here that move the toolhead. The default is to
#   not run any special G-Code commands on deactivation.

# Example settings to add to stepper_z section

# The safe_z_home section modifies the default G28 behavior
[safe_z_home]
home_xy_position: 100, 100
speed: 30
z_hop: 5
z_hop_speed: 25

[bed_mesh]
mesh_min: 4, 10
mesh_max: 190, 180
probe_count: 4, 4

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.013333, 0.195833, 0.083333, -0.157500
#*# 	-0.011667, 0.171667, 0.125833, -0.073333
#*# 	0.027500, 0.190000, 0.158333, 0.014167
#*# 	0.067500, 0.245833, 0.253333, 0.120000
#*# tension = 0.2
#*# min_x = 4.0
#*# algo = lagrange
#*# y_count = 4
#*# mesh_y_pps = 2
#*# min_y = 10.0
#*# x_count = 4
#*# max_y = 179.98
#*# mesh_x_pps = 2
#*# max_x = 190.0
#*#
#*# [probe]
#*# z_offset = 0.999

**webcam.txt**
### Windows users: To edit this file use Notepad++, VSCode, Atom or SublimeText.
### Do not use Notepad or WordPad.

### MacOSX users: If you use Textedit to edit this file make sure to use 
### "plain text format" and "disable smart quotes" in "Textedit > Preferences"

### Configure which camera to use
#
# Available options are:
# - auto: tries first usb webcam, if that's not available tries raspi cam
# - usb: only tries usb webcam
# - raspi: only tries raspi cam
#
# Defaults to auto
#
camera="auto"

### Additional options to supply to MJPG Streamer for the USB camera
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options
#
# Defaults to a resolution of 640x480 px and a framerate of 10 fps
#
camera_usb_options="-r 1280x720 -f 15"

### Additional webcam devices known to cause problems with -f
#
# Apparently there a some devices out there that with the current 
# mjpg_streamer release do not support the -f parameter (for specifying 
# the capturing framerate) and will just refuse to output an image if it 
# is supplied.
#
# The webcam daemon will detect those devices by their USB Vendor and Product
# ID and remove the -f parameter from the options provided to mjpg_streamer.
#
# By default, this is done for the following devices:
#   Logitech C170 (046d:082b)
#   GEMBIRD (1908:2310)
#   Genius F100 (0458:708c)
#   Cubeternet GL-UPC822 UVC WebCam (1e4e:0102)
#
# Using the following option it is possible to add additional devices. If
# your webcam happens to show above symptoms, try determining your cam's
# vendor and product id via lsusb, activating the line below by removing # and 
# adding it, e.g. for two broken cameras "aabb:ccdd" and "aabb:eeff"
#
#   additional_brokenfps_usb_devices=("aabb:ccdd" "aabb:eeff")
#
#
additional_brokenfps_usb_devices=(0bda:2137)

### Additional options to supply to MJPG Streamer for the RasPi Cam
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options
#
# Defaults to 10fps
#
#camera_raspi_options="-fps 10"

### Configuration of camera HTTP output
#
# Usually you should NOT need to change this at all! Only touch if you
# know what you are doing and what the parameters mean.
#
# Below settings are used in the mjpg-streamer call like this:
#
#   -o "output_http.so -w $camera_http_webroot $camera_http_options"
#
# Current working directory is the mjpg-streamer base directory.
#
#camera_http_webroot="./www-mainsail"
#camera_http_options="-n"

### EXPERIMENTAL
# Support for different streamer types.
#
# Available options:
#   mjpeg [default] - stable MJPG-streamer
#camera_streamer=mjpeg

### What did you expect to happen

A small Window showing the livestram from the webcam on the Dashboard

### How to reproduce

Just opening Fluidd with webcam plugged in and working

### Additional information

[image]

matmen, 2022-04-09:
This is caused by fluidd not being able to access the webcam. What URL are you using to access it? Can you post a screenshot of your camera settings (in the fluidd UI)?

homuser2003, 2022-04-09:
[image]

Sure! Thanks for your help!

matmen, 2022-04-09:
Append a `/?action=stream` to the URL and make sure you can access it from your browser

homuser2003, 2022-04-09:
You made my day! Makes perfect sense once I thing about it :-)

Thanks!

[image]

matmen, 2022-04-09:
no problem!

closed 2022-04-09 by matmen