fluidd-core/fluidd #629: Please integrate Annex-Engineering's print time estimator

sumguysr, 2022-04-15:
### Is your feature request related to a problem? Please describe

Currently Fluidd seems to estimate print times in a very naive and inaccurate way based on percentage printed and the print times estimated by a slicer. It would be great if Klipper Estimator was integrated to evaluate gcode on the current configuration and klipper's kinematics.

### Describe the solution you'd like

Add the ability to call Klipper Estimator on any file in the files interface, and add the option to call it it automatically on any file either when it's uploaded or when it's printed.

https://github.com/Annex-Engineering/klipper_estimator

### Additional information

Klipper estimator is here, https://github.com/Annex-Engineering/klipper_estimator

matmen, 2022-04-15:
While the Klipper print time estimator could be integrated directly into Fluidd (using WASM), I think it makes much more sense to have a Moonraker plugin pre-process the files (it makes sense to offload this to the printer host so the browser doesn't get overloaded). I think it would be a great addition, but it needs upstream support first then.

matmen, 2022-08-10:
Upstream issue: https://github.com/Arksine/moonraker/issues/481

gauravmm, 2022-08-10:
There's a bunch of useful static GCode analysis that can also be done in the same Moonraker plugin. I'll report back in a couple of weeks.

Dreistein75, 2022-11-18:
are there any updates?

pedrolamas, 2022-11-18:
@Dreistein75 as far as I'm aware, there has been no changes in Moonraker that would allow this feature.

gauravmm, 2022-11-18:
I'm attempting the plugin route, but with a Python implementation of the plugin. The goal is to also offer more functionality (essentially static analysis tools)

pedrolamas, 2025-02-12:
Moonraker has just added support for KlipperEstimator, so we can now start work on the Fluidd integration!

PR #1596 (2025-02-14): feat: adds Moonraker analysis support
Moonraker added a new module called `analysis`, that serves as an interface for klipper_estimator.

If the module is detected, a new "Perform Time Analysis" context menu item for gcode files will be shown, allowing to request Moonraker to invoke the klipper_estimator on a file; upon return of this call, we then refresh the file metadata.

[image]

Resolves #629
closed 2025-02-14 by pedrolamas