fluidd-core/fluidd #632: Gcode preview won't load a file

mistrjirka, 2022-04-17:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

I clicked "Load current File" in gcode preview ẃidget in Fluidd V1.17.0 and the screen showed, that the file was being loaded. The loading screen disappeared but nothing was loaded. Browser console showed this error:

- In Firefox
     TypeError: i.a is undefined
       Uncaught (in promise) TypeError: i.a is undefined
    59e4 http://192.168.1.16/js/0.44886169.worker.js:9
    e http://192.168.1.16/js/0.44886169.worker.js:1
    <anonymous> http://192.168.1.16/js/0.44886169.worker.js:1
    <anonymous> http://192.168.1.16/js/0.44886169.worker.js:1
[serializers.js:15:29](webpack:///node_modules/threads/dist-esm/serializers.js)

- in Brave/Chromium
    worker.mjs:3 Uncaught TypeError: Cannot read properties of undefined (reading 'expose')
    at Module.59e4 (worker.mjs:3:37)
    at e (bootstrap:19:22)
    at bootstrap:83:10
    at 0.44886169.worker.js:1:923
  serializers.js:15 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'expose')
    at Module.59e4 (0.44886169.worker.js:9:88832)
    at e (0.44886169.worker.js:1:110)
    at 0.44886169.worker.js:1:908
    at 0.44886169.worker.js:1:923
   

### What did you expect to happen

Working Gcode preview

### How to reproduce

Load any file into Gcode preview in Fluidd 1.17.0

### Additional information

The gcode preview doesn´t work in any browser on any OS.

matmen, 2022-04-17:
Hey, thanks for your report.
I can't seem to reproduce this on Win64 in Chrome 100 or Firefox 99.
What Chromium/Firefox version are you using?

pedrolamas, 2022-04-21:
Please also provide the g-code file that caused this issue!

pedrolamas, 2022-04-22:
@mistrjirka you mentioned above you are using Fluidd 1.17.0, please upgrade to 1.17.2 as we fixed quite a few bugs on the g-code previewer since then!

matmen, 2022-04-26:
Hey @mistrjirka, does updating to 1.17.2 solve this issue?

pedrolamas, 2022-05-01:
Fluidd 1.18.0 has just been released and hopefully should fix these issues.

I'm closing this now, please feel free to re-open if you are still experiencing this on the latest version.

closed 2022-05-01 by pedrolamas