fluidd-core/fluidd #638: Why didn't the configuration save and restart?

Cloud2702, 2022-04-21:
### Is your feature request related to a problem? Please describe

In the past, there was a shortcut to save and restart at the top of the version configuration. It is very convenient to save and restart once you modify the parameter points. But now the version has not been saved and restarted, only one is saved. After saving, you have to go back to the system and restart FIRMWARE. It's too much trouble to click. . .

### Describe the solution you'd like

I hope to add a shortcut to save and restart at the top of the configuration

### Additional information

fluidd v1.17.2
klipper v0.10.0-376

Cloud2702, 2022-04-22:
Look, edit printer.cfg I don't know why I only have a save here, but I don't save it and restart it.

Cloud2702, 2022-04-22:
[image]
Look, edit printer.cfg I don't know why I only have a save here, but I don't save it and restart it.

pedrolamas, 2022-04-22:
The only reason I could see for this to happen is if the service is not called "klipper".

On the main window, click the vertical "..." on the top right, then expand "Services", take a screenshot of that and share it here please!

Here's an example of mine:

[image]

Cloud2702, 2022-04-22:
[image]
With two machines, is the screenshot like this?

pedrolamas, 2022-04-23:
That's exactly it, and confirms what I suspected: as you have 2 Klipper instances, we don't actually know which one to restart when you save the file, hence why you don't see the "Save and Restart" button.

Cloud2702, 2022-04-23:
So install a klipper is able to see, and then install two klipper is not visible, is the reason for saving the file? Can we also solve the problem of saving and restarting with N machines or klipper in the future?

pedrolamas, 2022-04-24:
Fluidd is just a web interface, so all we can do is what the server (Moonraker) allows us to do. When only a Klipper instance exists, we can safely restart it as it is a single instance, but if multiple exist, we have no way of knowing directly which one is that config file referring to.

In any case, the solution for this will have to come from Moonraker first.

Cloud2702, 2022-04-24:
ok, thanks for the reply

pedrolamas, 2022-11-05:
I'm reopening this as Moonraker now supplies the correct service names so we can now restart the correct service!

PR #940 (2022-11-05): feat: use service names provided by Moonraker
Moonraker now provides the instance ids for the Moonraker and Klipper services, so we can restart the correct services when the respective config files change!

Fixes #638
closed 2022-11-05 by pedrolamas