fluidd-core/fluidd #643: How can I add my own function in web page?

Terbinm, 2022-04-24:
[image]

if I can add my own function in web page?? and which file?

matmen, 2022-04-24:
Hi, what kind of function would you like to add?
For custom logic you will need to set up your own dev instance as described in https://docs.fluidd.xyz/development
The required files for the file browser are in `src/components/widgets/filesystem`

Terbinm, 2022-04-24:
The discord channel inside,I just use camera function and another server to
do it.

Terbinm, 2022-04-24:
The discord channel inside,I just use camera function and another server to do it. 
[image]

Thank a lot.

matmen, 2022-04-26:
This isn't something we want to support natively because of performance concerns. However, you are free to set up your own fluidd build and add an `<iframe>` onto the dashboard. Help for setting up a dev instance is described here: https://docs.fluidd.xyz/development

closed 2022-04-26 by matmen