fluidd-core/fluidd #644: PiP for video feed.

FWeynschenk, 2022-04-26:
### Is your feature request related to a problem? Please describe

I'd like to be able to use most browsers' picture in picture feature. This is currently not possible because the video doesn't use the <video> tag.

### Describe the solution you'd like

Use html5 <video> elements for the video feed.

### Describe alternatives you've considered

Microsoft powerToys' always-on-top in combination with the video on it's own tab. But the UX isn't great compared to PiP.

matmen, 2022-04-26:
This is already available if you choose IP Camera mode: https://docs.fluidd.xyz/features/cameras

FWeynschenk, 2022-04-26:
Must've read right over that. Thank you.

closed 2022-04-26 by FWeynschenk