fluidd-core/fluidd #650: Macros - default parameters are not populating when read from object i.e. printer['configfile']

camerony, 2022-04-29:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

The menu to populate the default parameter for a macro is not able to use values from objects.  It's showing the name of the object instead i.e. "printer.configfile.settings.printer.max_velocity"

### What did you expect to happen

The value should populate from the object i.e. printer.configfile.settings.printer.max_velocity should show the max velocity i.e. 300

### How to reproduce

`[gcode_macro TEST_SPEEDABX]
gcode:
    {% set max_velocity = printer.configfile.settings.printer.max_velocity %}
    {% set speed = params.SPEED|default(max_velocity)|int %}
    M118 {speed}`
[image]

pedrolamas, 2022-04-29:
That's just how Klipper macros work, as they are evaluated at runtime only!

You can however try something like this:

```
[gcode_macro TEST_SPEEDABX]
gcode:
    {% set max_velocity = printer.configfile.settings.printer.max_velocity %}
    {% set speed = params.SPEED %}
    {% set speed = speed|default(max_velocity)|int %}
    M118 {speed}
```

camerony, 2022-04-29:
Unfortunately that doesn't work.  I could have sworn this used to work.

[image]

pedrolamas, 2022-04-29:
That actually worked as I expected: what you show on the screenshot is the hint text, so the field is empty (just click on it)

camerony, 2022-04-29:
hmm, interesting.  are you sure it never worked?  maybe klipper change the way it evaluated.  Andrew Ellis' macro uses this "https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/macros/TEST_SPEED.cfg"

pedrolamas, 2022-04-29:
Ah, I think I now see what you mean: we improved the defaults parsing a few versions ago, so now it can actually parse this default and show it, but as this is a variable and not a string, it's actually showing the wrong value!

The solution for now is using it like I showed above, at least until I have the chance to check the regular expressions on this!

pedrolamas, 2022-04-29:
Issue caused by #566

PR #651 (2022-04-30): fix: improves macro params default values parsing
Tested with the following example:

```
[gcode_macro TEST]
gcode:
  {% set A = params.A | default(0, true) | float %}
  {% set B = params.B | int | default(0.005, true) | float %}
  {% set C = params.C | default("PLA", true) %}
  {% set D = params.D | default("PLA") %}
  {% set E = params.E | int | default(0.2) %}
  {% set F = params.F|default(printer.configfile.settings.printer.max_velocity)|int %}
  {% set G = params.G|default("this \"works\"")|int %}
```

Before (F has default value when should not):

[image]

After (no default value on F):
 [... cut]
closed 2022-04-30 by matmen