fluidd-core/fluidd #656: Render MD files

digitalninja-ro, 2022-05-06:
### Is your feature request related to a problem? Please describe

The Docs tab from Configuration shows only raw sources for .md files. It's nice to have all the documentation at a click distance but it's not so human readable.

### Describe the solution you'd like

Will be more handy to render the md files in fluidd in case I want a quick reference or don't have an internet connection to access the nice formatted documentation from klipper website or git repo.

pedrolamas, 2022-05-16:
I've given a quick try on this and actually got it to render markdown, but this made me think that we probably should instead just link directly the open document file to the Klipper website instead... that way we ensure that users are looking at the latest documentation available!

digitalninja-ro, 2022-05-16:
Thank you for your answer but I think the documentation is also updated and pushed with regular commits. If an user has an older version of klipper on pi  he also has the local documentation for his version of klipper.   It's not hard to have a tab open with klipper website or git but as long as we have the documentation on local machine how difficult is to make it more human readable?

pedrolamas, 2022-05-18:
Well, it is possible... though at the moment, links don't work, images don't work...

[image]

PR #1376 (2024-02-27): feat: view rendered Markdown files
Allows viewing of rendered Markdown (*.md and *.markdown) files.

## Context menu for Markdown files

[image]

## Markdown viewer desktop

[image]

## Markdown viewer mobile

[image]

Resolves #656
closed 2024-02-27 by pedrolamas