fluidd-core/fluidd #658: Gcode preview parsing issues with negative values and no leading zeros

LucidWolf, 2022-05-06:
### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

Gcode preview has incorrect visualization of path.  
Expect due to Cura output not providing proceeding zeros.  Issue happens what appears to be sporadically but I am guessing it happens because of a following G0 or when crossing the x or y axis sometimes.  Looks to be something wrong in the parsing.

I uploaded 2 files one of a delta calibration with no fix to the gcode where i believe the preview fails at line 609 in the file.  I have included a second gcode where i did a replace of -. with -0. and no issues are found in the preview.

Ex.)
G1 X5.448 Y-.259 E329.2888

[image]
VS
G1 X5.448 Y-0.259 E329.2888
[image]

### What did you expect to happen

Gcode preview to match gcode.  Works if using the provided file MD1_06_calibrate_size_fix.gcod
[MD1_06_calibrate_size_fix.gcode.txt](https://github.com/fluidd-core/fluidd/files/8642214/MD1_06_calibrate_size_fix.gcode.txt)
e

### How to reproduce

Use the provided MD1_06_calibrate_size.gcode
[MD1_06_calibrate_size.gcode.txt](https://github.com/fluidd-core/fluidd/files/8642216/MD1_06_calibrate_size.gcode.txt)

pedrolamas, 2022-05-26:
@LucidWolf The issue with decimals without leading 0 (so something like ".12") was fixed in the latest 1.18.1, can you please update to it and confirm if you can still reproduce this issue? Thank you!

LucidWolf, 2022-05-27:
Concur that issue is fixed in 1.18.1.  Thanks for the quick turn around.

closed 2022-05-27 by LucidWolf