fluidd-core/fluidd #662: Jobs Does Not Display Print Duration or Total Duration, but History Does

Imburr, 2022-05-08:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

When I access the Jobs section on the home page, or the Jobs sidebar page, Print Duration and Total Duration are empty. When I visit History, they show correctly.

Home Page Jobs: https://i.imgur.com/r49ltdW.png
Jobs Page: https://i.imgur.com/QBRbWYH.png
History: https://i.imgur.com/Wh3clmU.png

### What did you expect to happen

I expect PD and TD to show in jobs sections.

### How to reproduce

Confirmed with another user on Reddit that they have the same issue.

### Additional information

Willing to post whatever necessary in terms of logs/code, let me know.

Imburr, 2022-05-08:
Some discussion here: https://www.reddit.com/r/klippers/comments/ukk855/no_printtotal_duration_in_fluidd_on_klipper/

matmen, 2022-05-10:
Hey, thanks for your bug report!

I've tracked this down to be a race condition in [@/store/files/actions.ts](https://github.com/fluidd-core/fluidd/blob/develop/src/store/files/actions.ts#L57-L60)
Basically, whenever the files list comes in before the history (after a full refresh of fluidd), the table is left without job history metadata. An easy fix for that would be to just wait for history metadata (if history is supported). I'll work on a fix in the next couple days.

Imburr, 2022-05-10:
> Hey, thanks for your bug report!
>
> I've tracked this down to be a race condition in [@/store/files/actions.ts](https://github.com/fluidd-core/fluidd/blob/develop/src/store/files/actions.ts#L57-L60) [...]

Thank you for all of your hard work! Great product!!

PR #668 (2022-05-15): fix: job card history metadata race condition
Closes #662
closed 2022-05-15 by matmen