fluidd-core/fluidd #669: Cannot start print

MooNag, 2022-05-13:
### Browser

Firefox

### Device

Desktop PC

### Operating System

Linux

### What happened

I updated my Voron printer configured with fluidd. It was working right before the update. Last time I updated my printer was probably about a month or two ago.

These are the versions I updated to and the printer has been restarted:

fluidd
v1.18.0
UP TO DATE
klipper
v0.10.0-408-g2c441b45
UP TO DATE
moonraker
v0.7.1-569-gbabfba1
UP TO DATE
OS Packages
UP TO DATE 

I followed this guide to move some config options into different headings:
[https://moonraker.readthedocs.io/en/latest/configuration/#option-moved-deprecations](https://moonraker.readthedocs.io/en/latest/configuration/#option-moved-deprecations)

Reverting the config changes does not stop the issue

No errors are showing in the GUI

### What did you expect to happen

I expect the print to start

### How to reproduce

Attempting to start a print does nothing. Every other printer control function seems to work properly (home, jog, temp, etc)

### Additional information

Relevant logs

fluid-error.log
```
2022/05/12 22:36:27 [error] 675#675: *51 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.2.11, server: _, request: "GET /access/oneshot_token HTTP/1.1", upstream: "http://127.0.0.1:7125/access/oneshot_token", host: "192.168.2.53", referrer: "http://192.168.2.53/"
```

moonraker.log
```
2022-05-12 22:53:26,167 [websockets.py:build_error()] - JSON-RPC Request Error: -31000
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 253, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 364, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 75, in _gcode_start_print
    return await self.start_print(filename)
  File "/home/pi/moonraker/moonraker/components/klippy_apis.py", line 118, in start_print
    await self.klippy.wait_started()
  File "/home/pi/moonraker/moonraker/klippy_connection.py", line 116, in wait_started
    asyncio.shield(self.connection_task), timeout=timeout)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

matmen, 2022-05-13:
The `Connection refused` seems to be unrelated here, likely caused by a moonraker restart.
For the JSON-RPC error: It looks like something changed in the communication between Klipper and moonraker. You'll need to wait for a fix on moonraker's (or Klipper's) side here I'm afraid.
For reference: https://github.com/Klipper3d/klipper/issues/5500

matmen, 2022-05-13:
Actually, this does seem kinda odd. The klippy log file in the linked issue shows Klipper is actually processing (/printing) a file, so I don't see why the print wouldn't be running. Can you try running `SDCARD_PRINT_FILE FILE=<filename>` in the console?

matmen, 2022-05-13:
Fixed in https://github.com/Arksine/moonraker/commit/0164bb239ea1032e105428435ff2643378bf4bfa