fluidd-core/fluidd #670: Temperature limit wrong

gfarcas, 2022-05-13:
### Browser

Chrome

### Device

Desktop PC

### Operating System

macOS

### What happened

Fluidd shows temperature limit active, but the system show's a different temp
[image]

### What did you expect to happen

To show the system temp

### How to reproduce

I don't know

matmen, 2022-05-13:
The notification shows the temperature the limit first triggered at, so this is intended behavior. I don't think it's worth refactoring the notification system just to show an up-to-date temperature (and you can already [show the current temperature in the temperature graph](https://www.klipper3d.org/Config_Reference.html#host-temperature-sensor), too). @pedrolamas Thoughts?

gfarcas, 2022-05-13:
> The notification shows the temperature the limit first triggered at, so this is intended behavior. I don't think it's worth refactoring the notification system just to show an up-to-date temperature [...]

I have checked the temp seconds after the notification, I don't think it dropped 10 degrees so fast

matmen, 2022-05-13:
Your Pi must have reached 80°C for a short time (assuming you're running a RPi 4), the throttling warning only gets triggered >80C by the Pi directly (the status is retrieved via vcgencmd). It probably spiked at 81C and then went down again.

matmen, 2022-05-13:
For reference: The throttle state is read by moonraker by running `vcgencmd get_throttled`, the core temperature is read from the `/sys/class/thermal/thermal_zone0/temp` file.

matmen, 2022-05-20:
Is this still an ongoing issue or would you consider this solved?

closed 2022-05-28 by gfarcas