fluidd-core/fluidd #675: Using two machines at the same time will report an error

Cloud2702, 2022-05-15:
### Browser

Edge

### Device

Desktop PC

### Operating System

Windows

### What happened

[image]
[image]

Two machines can not be used at the same time, such as A machine in use, and then switch another B machine has an error, even after restarting FIRMWARE A machine will report an error (Failed automated reset of MCU 'opi')

### What did you expect to happen

Hope to solve this problem of using multiple machines at the same time

### How to reproduce

No idea

### Additional information

fluiddv1.17.2
klipperv0.10.0-431

matmen, 2022-05-15:
I don't think this is a fluidd bug - the warnings shown are just directly from Klipper. I assume you have misconfigured one of the printers (maybe you're using the same device ID for both printers? What's your `serial` value in the `[mcu]` block of both of your `printer.cfg` files?). You're probably better off [asking the Klipper community](https://www.klipper3d.org/Contact.html) for help.

matmen, 2022-05-20:
Are you still facing this issue or were you able to solve it?

Cloud2702, 2022-05-20:
Has been solved to find out the adxl345 accelerometer opi problem

closed 2022-05-20 by Cloud2702