fluidd-core/fluidd #681: New slider (PA, smooth time) does not work proper with extruder_stepper config

homasholz, 2022-05-18:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

when using the new sliders, the command behind it does not work with my extruder config

[image]
[image]
[image]
[image]
[image]

### What did you expect to happen

that the sider/command works proper

### How to reproduce

using extruder_stepper in config

pedrolamas, 2022-05-18:
@homasholz Thank you for taking the time to submit this bug report!

I've confirmed this issue, seems we need to add support for `extruder_stepper` configurations which we currently don't have.

pedrolamas, 2022-11-04:
Granted that I might be wrong here, but from my tests it seems that `SET_PRESSURE_ADVANCE` now works with multiple `extruder_stepper` entries (though it just sets the active `extruder`, not the `extruder_stepper`!

At least Fluidd 1.20.1 is now showing the Pressure Advance controls (though we didn't actually change anything!) and these seem to work... are you able to confirm this, @homasholz ?

homasholz, 2022-11-04:
I dont have the sliders available. 
[image]

pedrolamas, 2022-11-04:
Are you using the latest (dev) version of Klipper?

homasholz, 2022-11-04:
i am on v0.10.0-615-g96ea871b 

[image]

pedrolamas, 2022-11-04:
Ah, there you go then! A have a feeling you will need to press that little "update" button next to Klipper to get that on working condition.

Be aware that if you update Moonraker you will also need to update the folder structure (it should do it automatically, but I've seen reports of people having to do extra manual steps)

homasholz, 2022-11-04:
Ok. I have updated everything.
But no PA sliders appear (i think they should be in tools section).

pedrolamas, 2022-11-04:
Maybe I was wrong then... Nevertheless, you might want to try and manually send a `SET_PRESSURE_ADVANCE ADVANCE=0.5` and see what happens!

homasholz, 2022-11-04:
[image]

pedrolamas, 2022-11-05:
If you are able to upload a `klippy.log` from this printer so I can take a look I would be very appreciated!

homasholz, 2022-11-05:
[klippy.log.txt](https://github.com/fluidd-core/fluidd/files/9944481/klippy.log.txt)

pedrolamas, 2023-02-21:
@homasholz FYI, we've finally fixed this issue on Fluidd 1.23.2!

You will now see pressure advance controls per `extruder_stepper` (#1022)

If you also update to the latest Klipper available (which I strongly recommend), you will also get options to view/change the synced extruder (#1034) and enable/disable the stepper (#1039)

I'd be really interested in your feedback on these features! 🙂

homasholz, 2023-02-25:
GREAT!!
everything is working as expected with my setup. 

[image]

pedrolamas, 2023-02-25:
Thanks for confirming @homasholz , took a while but we got there in the end! 😁

PR #694 (2022-05-26): fix: hide Pressure Advance if no default extruder stepper set
Klipper doesn't seem to provide what `extruder_stepper` is currently active so Moonraker can't make it available to the API consumers.

The only way forward in this for now is to just "hide" the full Pressure Advance controls if no `pressure_advance` values are provided in the active extruder (meaning no stepper is set)

Fixes #681
PR #1022 (2023-02-02): feat: Pressure Advance per extruder stepper
The current Pressure Advance controls work fine when there is an `[extruder]` with a stepper attached, but when we have them separately (by using `[extruder_stepper]` entries), this would just not work as there is no "active extruder stepper" entity)

To fix this, we will now show the existing PA controls for active extruder if it has a stepper attached, and a new separate box with PA controls for each of the separate extruder steppers.

[image]

These boxes are collapsible and by default they will load in collapsed state:

[image]

(Note: one can ea [... cut]