fluidd-core/fluidd #684: G code preview with large number of null layers

yasaslive, 2022-05-18:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

- After updating to new fluidd G code previewer shows large number of null layers. Only last layers have the paths. 

### What did you expect to happen

Shows correct layers and layer path

### How to reproduce

Update to latest fluidd
[image]
[image]

pedrolamas, 2022-05-18:
@yasaslive thanks you for submitting the bug report.

Can I ask that you supply a sample g-code file so we can check? Thank you!

yasaslive, 2022-05-18:
This is one of my gcode that generated couple minutes ago with PrusaSlicer 2.4.2. 
Infill: 40%
Layer height: .1mm
Speed: 100mm/s
Material: ABS

[money_cat.txt](https://github.com/fluidd-core/fluidd/files/8720452/money_cat.txt)

pedrolamas, 2022-05-18:
Duplicate of #674

Good news: this was fixed by #679 yesterday and the fix has already been merged to the `develop` branch!

Here's a screen-capture of the file you provided showing it works on the latest `develop`:

[image]

You can either check the `develop` branch or way for 1.18.1 release that should be available soon!

pedrolamas, 2022-05-18:
Duplicate of #674

closed 2022-05-18 by pedrolamas