fluidd-core/fluidd #690: Move the g-code preview view options to drop down settings button like other panels

alexrmay91, 2022-05-22:
### Is your feature request related to a problem? Please describe

[image]

I keep Fluidd open on my second monitor, which is in a vertical/portrait mode. This causes some issues with how Fluidd is rendered, like an extremely small g-code preview window.

### Describe the solution you'd like

[image]

If the view options were moved to a drop down settings button (like the console panel), there would be a lot more space for a larger g-code preview window.

### Describe alternatives you've considered

None.

pedrolamas, 2022-05-24:
Fair point, makes complete sense and it is way more consistent with the rest!

I think we can take a look at this once #693 gets merged!

alexrmay91, 2022-05-24:
You're on top of it! Much appreciated

PR #696 (2022-05-26): fix(layout): move gcode controls to dropdown; fix vertical layout
Moves the gcode preview related settings to a dropdown and fixes the layout to be compatible with vertical monitors
[image]
[image]

Not sure if the "Load current file" and "Clear file" buttons should also go there, LMK what you think..

Closes #690