fluidd-core/fluidd #697: Klipper and moonraker invalid, unable to recover!

hamed1998googoogaga, 2022-05-26:
### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

I tried to print on a Raspberry Pi 3 Model B Rev 1.2 but could not because the Z was triggered, calibration had failed

After looking around I saw that both klipper and moonraker are "invalid" in the settings page and when I pressed recover it failed to do so.

The OS packages are also needing updates, but when I pressed those it failed aswell
[image]
.
[image]

That last pic is me trying to recover klipper, moonraker is roughly the same, here is what happens when I try to update OS packages:
[image]

### What did you expect to happen

I expected fluidd to recover both klipper and moonraker, and update the OS packages, but nothing happened

### How to reproduce

Honestly I have no idea, Im very new to 3d printing and kinda just opened up the settings page to see this

### Additional information

Im not sure how to edit any proxy or code or anything, I tried looking around the gcode file are but Im stumped as to what Im looking at. Sorry in advance for my incompetence!

pedrolamas, 2022-05-26:
Hi @hamed1998googoogaga this means your Raspberry Pi can't access GitHub, so this is either a network issue or GitHub was down when it tried to access it.

I would recommend rebooting the Pi and trying again at a later time.

hamed1998googoogaga, 2022-05-26:
Ok I will try this tomorrow then and will let you know!

hamed1998googoogaga, 2022-05-27:
@pedrolamas Is there anyway to know when GitHub is down? I tried again today and still the same issues, my network seems to work fine.

Thanks!

hamed1998googoogaga, 2022-05-27:
^^^^Nevermind, I think I found out where to put the code, but I just need to actually connect the ssh to the pi, it is telling me that the host does not exist for some reason, Ill try to figure out why and get back to you

hamed1998googoogaga, 2022-05-27:
Hey @pedrolamas here is what happens when I insert that code into there
[image]

pedrolamas, 2022-05-27:
That confirms that you do have a network issue as you can't reach github.com

You can also try `curl https://1.1.1.1`, if this works then the problem is most likely a DNS server issue.

Not much I can say besides telling you to check your firewall and DNS server...

hamed1998googoogaga, 2022-05-30:
Thanks for everything, I think you are right, it may be a network error, since I dont have a DNS server and I tried with my firewall turned off.

pedrolamas, 2022-06-03:
@hamed1998googoogaga, given this, are you ok for us to close this issue?

hamed1998googoogaga, 2022-06-03:
@pedrolamas yeah for sure

closed 2022-06-03 by matmen