fluidd-core/fluidd #698: Problems with a dashboard when using -awb key

HaroldSP, 2022-05-27:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Hello!

I have a NOIR raspberry camera, so normally it streams in shades of pink.
To fix that (when you don't need it to stream at night) on a regular Pi OS I use that command with -awb greyworld key:
`raspivid -p 0,0,640,480 -fps 30 -t 0 -awb greyworld --fullscreen`

So in the webcam.txt file I made that line:
`camera_usb_options="-r 640x480 -f 30 -awb greyworld"`
And this is the result:
[image]

At the beginning I thought that the video streams but the window is collapsed, but actually, there is no stream.
Openning in a full screen mode gives a 502 Bad Gateway.

After removing -awb greyworld key, it works fine:
[image]

I have also tried to use other -awb key options (https://community.octoprint.org/t/available-mjpg-streamer-configuration-options/1106), still the same issue.
So is there any idea how to fix that.

Thank you very much.

### What did you expect to happen

I expect to see a regular stream with -awb greyworld key (IR filter).

### How to reproduce

Just try to insert -awb key, for example:
`camera_usb_options="-r 640x480 -f 30 -awb greyworld"`

matmen, 2022-05-27:
The `awb` option isn't supported by the UVC plugin. I assume you're running the raspicam in USB mode? You will need to run it in `camera="raspi"`-mode for raspicam options to work.

HaroldSP, 2022-05-29:
> The `awb` option isn't supported by the UVC plugin. I assume you're running the raspicam in USB mode? You will need to run it in `camera="raspi"`-mode for raspicam options to work.

Thanks for your reply.

I did that before, and there was no video feed at all.

It was like this:
`camera="raspi"`

`camera_raspi_options="-r 640x480 -f 30 -awb greyworld"`

More than that, it neither works even without -awb key:
`camera_raspi_options="-r 640x480 -f 30 -awb greyworld"`

So maybe there's a problem in **### Configuration of camera HTTP output** section?

matmen, 2022-05-29:
I see, thanks for your feedback. I can't troubleshoot this (as I don't have a Pi camera), so we will have to wait for someone who has one to chime in ;)