fluidd-core/fluidd #727: "Invalid root path (gcodes)" on initial launch

pedrolamas, 2022-06-23:
### Browser

Edge

### Device

Desktop PC

### Operating System

Windows

### What happened

If no printer.cfg file is found on first launch, Fluidd will display an "Invalid root path (gcodes)" error message:

[image]

### What did you expect to happen

Error message should not be shown

### How to reproduce

- Stop Klipper and Moonraker services
- Delete/rename `~/.moonraker_database`
- Delete/rename `~/klipper_config/printer.cfg`
- Start Klipper and Moonraker services
- Open Fluidd

pedrolamas, 2022-06-23:
The easy mitigating of this problem is to just have a `printer.cfg` file before trying to start Klipper!

However, on further analysis I found that this issue only started with the changes from 5ce0921c25d306fa651c5b05f1537829d56d5e67, so this is where we need to fix!

Leon-Du, 2022-06-23:
hello, I tried this method to solve the problem of "inviad root path gcodes" by your method above, but I can`t deal with it, could you help me to check this problem, thank you very much.
[klippy.log](https://github.com/fluidd-core/fluidd/files/8967956/klippy.log)

pedrolamas, 2022-06-23:
Hi @Leon-Du, if you open the log and scroll all the way down, you will find the error there:

> Error: Option 'sensor_type' in section 'heater_bed' must be specified

Your `heater_bed` config is missing the `sensor_type`, once you add that in, it should be good!

Check here the full config: https://www.klipper3d.org/Config_Reference.html#heater_bed

Leon-Du, 2022-06-23:
> Hi @Leon-Du, if you open the log and scroll all the way down, you will find the error there:
>
>
> Your `heater_bed` config is missing the `sensor_type`, once you add that in, it should be good!
>
> Check [...]

hi Pedro lamas, thank you for your message,  I will try to deal with the error by your suggestion. I am a green hand,  I hope I can slove it, Lol

pedrolamas, 2022-06-23:
@Leon-Du my advice is that you come by the Klipper Discord where you can always find people ready to help in almost real-time: https://www.klipper3d.org/Contact.html#discord-chat

PR #728 (2022-06-23): fix: check for root available before listing files
Check that `gcodes` root exists before attempting to list the files from it.

Fixes #727