fluidd-core/fluidd #740: Default LED section is not added to UI

Smoria, 2022-07-04:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

When `[led]` section is added to Klipper config Fluidd doesn't create output UI item for it.

Klipper config:
```cfg
[fan]
pin: gpio20

[fan_generic Driver]
pin: gpio17

[led Headlight]
white_pin: gpio18
initial_WHITE: 0.0
```

Here is the result UI:
[image]

If I would change `[led]` section to `[neopixel]` it's configured properly. As seems by the code `led` section are recognized as valid LED output item, but it's trying to access invalid config items like `color_order`. Same issue for full list of color pins:
```cfg
[led Headlight]
white_pin: gpio18
red_pin: gpio21
green_pin: gpio23
blue_pin: gpio26
initial_WHITE: 0.0
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 0.0
```

### What did you expect to happen

LED output UI item with only white color must be added to `Fans & Outputs` block.

### How to reproduce

Use provided config to reproduce the issue

Smoria, 2022-07-19:
@matmen could you please re-open issue. It's still persists in 1.19.0. Same steps to reproduce.

pedrolamas, 2022-07-19:
Back to the drawing board....

closed 2022-07-19 by pedrolamas