fluidd-core/fluidd #741: Wrong documentation linked in the config editor

aka13-404, 2022-07-05:
### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

The config manager tries to link sections of the telegram.conf to the klipper documentation. 
[image]

### What did you expect to happen

I would expect it to not link anywhere.
Ideally I would like for fluidd to link to our official documentation, https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#bot for example. 

### How to reproduce

Add a telegram.conf file with any section as on the screenshot.

pedrolamas, 2022-07-05:
@nlef are we ok to link to `https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#[section_name]` or should we just disable Codelens support for telegram.conf?

nlef, 2022-07-05:
> @nlef are we ok to link to `https://github.com/nlef/moonraker-telegram-bot/wiki/Sample-config#[section_name]` or should we just disable Codelens support for telegram.conf?

This is a valid config url. Codelens will help users a lot, so it should not be disabled.
Thanks.

closed 2022-07-05 by pedrolamas

aka13-404, 2022-07-05:
You are a magician Pedro, what would we do without you!

PR #742 (2022-07-05): fix: codelens links for moonraker-telegram-bot
Refactors Monaco setup to use existing configuration map and to expose Codelens support for moonraker-telegram-bot.

Fixes #741