fluidd-core/fluidd #753: Displaying Printers as Tabs on Top of the Overview

XoBoR, 2022-07-08:
### Is your feature request related to a problem? Please describe

I'm running a lot of printers on only a few Host computers running Klipper / Moonraker / Fluidd. Having to switch to each individual Printer Via the Menu in the top right in order to see what it's currently doing doesn't seem optimal to me. 

### Describe the solution you'd like

[image]

I made a quick drawing to illustrate what I have been thinking about.

Having all the Printers status always in view and being able to switch to them without going through the Menu seems like a good Idea to me.

I'm not sure how hard this would be to implement or if it makes a lot of sense to others users.

Thanks for the great Fluidd anyways and have a good day!

matmen, 2022-07-11:
Related: #292

ChaosBlades, 2022-10-11:
I think it would be a good idea to implement https://github.com/fluidd-core/fluidd/issues/464 the same time this gets implemented.