fluidd-core/fluidd #760: Подключение 2 принтеров к Raspberry pi

Fiilliin, 2022-07-12:
### Is your feature request related to a problem? Please describe

С помощью KIAUH установил 2 комплекта клиппера, мунракера и флюида. 
[image]
При установке присвоил принтерам имена (anet и ebot). открываю браузер, ввожу 192.168.1.41/anet или 192.168.1.41/ebot - результат одниаков - я постоянно нахожусь на одном принтере, на второй он не переключается. Подскажите, что я делаю не правильно и как это починить?

### Describe the solution you'd like

Управление обоими принтерами из браузера

matmen, 2022-07-12:
You can add another printer using the side menu, accessible via the 3 dots in the top right corner. There's a button to add another printer, as the URL you will need to set set `192.168.1.41:[port]`, replacing `[port]` with whatever port you set in your 2nd instances moonraker configuration.
Let me know if you need additional support 😉

Fiilliin, 2022-07-12:
Есть ли возможность управлять одновременно двумя принтерами? т.е. открыть 2 вкладки браузера - на одной один принтер, на другой второй принтер.

matmen, 2022-07-12:
It's not really supported as of now (reloading the 1st printers page would set it to use the 2nd printer), but it should still work if you open two tabs and switch to the other printer on the 2nd tab. Alternatively you can open an incognito browser window for the sessions to not collide.
Proper support for multi-printer management is already planned.

closed 2022-07-12 by Fiilliin