fluidd-core/fluidd #761: UI for manual bed levelling

rsheasby, 2022-07-13:
### Is your feature request related to a problem? Please describe

I recently got an Ender 3 S1 Pro, and unfortunately got unlucky with a misbehaving CRTouch. Repeatability is fine, but there's probe location bias that I haven't been able to track down. No major problem though, as I can create a manual mesh that works pretty damn well. However, the lack of a UI wizard or utility makes this really annoying to deal with in fluidd. Manually typing in the TESTZ commands is painful and makes the whole process much slower.

### Describe the solution you'd like

A simple UI that provides easy and quick methods for starting a manual mesh level and issuing the testz commands to find the right height.

It doesn't need to be very complex, just a simple layout to start the process, then 6 buttons to move the head up and down by various amounts (1mm, 0.1mm, 0.01mm). I've found what works well for me is going down 1mm at a time until the nozzle touches the paper, then going up 0.1mm at a time until the paper moves freely, then going down 0.01mm until the paper touches again.

### Describe alternatives you've considered

Creating macros for issuing the correct commands does work, but the layout is cumbersome and makes it slower to work through a larger mesh. Ideally the UI would be as easy as possible to encourage people without ABL (or with broken ABL) to actually create and update a bed mesh instead of relying purely on corner levelling.

pedrolamas, 2022-07-22:
Related to #782

pedrolamas, 2022-07-22:
Adding a few options for layout here...

## Option 1

[image]

## Option 2

[image]

@matmen @BastelKlug thoughts?

BastelKlug, 2022-07-22:
I like the second option more. But shouldn't you replace the - and + icon with this:
[image]

pedrolamas, 2022-07-22:
A good point... the only reason I went with `-` and `+` was to show exactly what it will send on `TESTZ Z=*`

matmen, 2022-07-22:
I think option 3 looks the best, but maybe the buttons groups can line up? 
[image]
[image]

As for the icon, I think both +/- or the material design icons would work..

PR #785 (2022-07-23): feat: adds new Manual Probe helper dialog
Will show buttons for all the "Z Adjust values" from settings + `1` and `0.1` in case they are not there as this might need more broad changes.

There's a new "Show Manual Probe dialog automatically" setting that will automatically open the dialog if we are on the Dashboard or Tune page and a Manual Probe command is initiated.

Dialog:

[image]

Button added to dashboard page (only visible when a Manual Probe command is running):

[image]

Button added to tune page (only visible when a Manual Probe command is running):

[image]

Fixes #761
Fixes #782
closed 2022-07-23 by pedrolamas