fluidd-core/fluidd #764: I don't think Fluidd knows it is on v1.19

Majawat, 2022-07-13:
### Fluidd Version

v1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

Fluidd announced there was an update for Fluidd, Klipper and KlipperScreen. I updated the latter two, but trying to update Fluidd gives me these messages:

```
Updates finished
Updating Web Client fluidd...  
Downloading Client: fluidd
Error updating fluidd
```

I also updated manually via https://docs.fluidd.xyz/updates/manual#fluidd thinking it didn't update correctly.

However, in writing up this bug report, I looked at the .version file in ~/fluidd and it shows v1.19.0. So now I think that fluidd doesn't know it's on v1.19.0. To double check, I compared the release notes with what's installed. I do see the the Codelens, Save & Restart, and optional 4 column view.

### What did you expect to happen

Fluidd knows what version its on? Or at least updates correctly if it hasn't.

### How to reproduce

Update from v1.18.2 to v1.19.

### Additional information

[image]

pedrolamas, 2022-07-13:
Fluidd doesn't actually do any update as it is just a webclient running on the browser; Moonraker is the one doing the update here!

Most likely this is an issue with something on moonraker.conf, can you please upload you file here (or copy & paste the contents of the file to a comment)

pedrolamas, 2022-07-13:
Another thing you can also check is what Moonraker returns from `http://<your-moonraker-host>:7125/machine/update/status?refresh=false` (just replace `<your-moonraker-host>` here with your Moonraker hostname or IP address)

Majawat, 2022-07-13:
Ah, found it.

```
# this enabled fluidd updates
[update_manager client fluidd]
type: web
repo: fluidd-core/FluiddPi
path: ~/fluidd
```

That should be `fluidd-core/fluidd`. Not sure why `fluidd-core/FluiddPi` was there. Removing the `Pi` resolved this issue. Thanks and sorry to bother!

closed 2022-07-13 by Majawat