fluidd-core/fluidd #765: After changing Neopixel color, it keeps flipping the red and green values over and over.

Majawat, 2022-07-13:
### Fluidd Version

1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I change the light wheel to a color, and the console keeps flipping the G and R values (leaving B alone). This flashes the lights between two colors until it finally stops.

### What did you expect to happen

It to stay on the chosen color

### How to reproduce

Change neopixel color via color wheel or console.

### Additional information

Here's my neopixel config:

```
[neopixel Bed_Lights]
pin: PE6
chain_count: 5 ;number of pixels on the strip
color_order: GRB ;The default is GRB.
#initial LED color
initial_RED: 0.25
initial_GREEN: 0.25
initial_BLUE: 0.25
```

I changed the color to red on the color wheel and this is the console:
```
17:33:52 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:33:53 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:33:55 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:33:56 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:33:58 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:33:59 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:00 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:01 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:02 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:03 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:04 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:05 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:06 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:07 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:08 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:09 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:10 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:11 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:12 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:14 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:15 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:16 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:17 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:18 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:19 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:20 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:21 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:22 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:23 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:24 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:25 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:26 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:27 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:28 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:29 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:30 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:31 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:32 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:33 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:34 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:35 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:36 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:37 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
17:34:38 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=0 undefined=0
17:34:39 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=0 undefined=0
```

Where it finally stayed on red. This also occurred when I manually set the color to purple with the console:

```
17:39:41 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=1 (<--manual command)
17:39:42 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=1 undefined=0
17:39:43 $ SET_LED LED=Bed_Lights GREEN=1 RED=0 BLUE=1 undefined=0
17:39:44 $ SET_LED LED=Bed_Lights GREEN=0 RED=1 BLUE=1 undefined=0
...etc.
```

pedrolamas, 2022-07-13:
I've managed to confirm there are some issues here, (especially with the `undefined` last bit of the command), so will have to make some further tests to fix this problem.

pedrolamas, 2022-07-15:
FYI, I managed to reproduce this issue and now have a PR with some changes that I believe will fix the problem completely!

PR #766 (2022-07-15): fix: OutputLed working with all types of LEDs
A couple of issues with this one, but I've now tested in Simulavr with `[led]`, `[neopixel]` and `[dotstar]` (containing 1 or more channels, with and without White channel), and I believe it will work for these and the other types of LEDs too!

> **Dev note:** though on some of these LED types we can specify a `color_order`, that does NOT change the output order of `color_data` - it is ALWAYS `[R, G, B, W]`)
> 
> [image]
> 
> Klipper also does not complain if we call `SET_LED` with all params, so I've decided to keep that in for now

Fixes #765