fluidd-core/fluidd #769: After recent update I get a save config and restart just from a print with no other config changes

sprior, 2022-07-16:
### Fluidd Version

v1.19.0-6184c73

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I like the recent changes to Fluidd, but now when I just turn the Voron on and run a print job a banner appears up top and then becomes enabled to save config and restart even when I have made no config changes, so I'm not sure what the difference would be.  As part of my print_start macro I do QGL, auto-z-calibration, and bed mesh (which gets cleared at the end of the print), so I don't know if one of those things is getting picked up as a config change to save, but I don't think it should be.

### What did you expect to happen

I don't think that just printing something should trigger an enabled save config and restart banner item.

### How to reproduce

I think I covered this in "What Happened"

matmen, 2022-07-16:
I assume this would be because of bed mesh calibration, but I'm sure we can add an option to toggle the button off 😉

pedrolamas, 2022-07-16:
@sprior out of curiosity here, but if you enable the "Review Pending Configuration Changes before Save and Restart" setting and then click on the "Save config and restart" button, what do you get there? Can you share a screenshot please?

sprior, 2022-07-17:
@matmen @pedrolamas  I checked and it is bed mesh. 
[image]

pedrolamas, 2022-07-17:
> I checked and it is bed mesh.

I guess a quick fix right now would probably be to add a `BED_MESH_CLEAR` to your "end print" macro!

matmen, 2022-07-17:
That would still show the button during printing.. I'm fine with adding a toggle for the button here, though of course that wouldn't just hide the button completely instead of just for bed mesh changes..

sprior, 2022-07-17:
@matmen I think I'd agree with a toggle to hide this button completely - for me when I'm actually making config changes that's something I'm doing very deliberately so I seek out the button to save those changes and when I'm not doing that I'd be super careful before saving config changes I didn't do very carefully.

pedrolamas, 2022-07-17:
I see 3 ways of moving here:

- add toggle that would work as such:
  - never show **OR**
  - show when has changes
- add 3 state option (dropdown), with the following:
  - never show **OR**
  - show when has changes, but not during busy/printing **OR**
  - show when has changes
- do nothing and assume current behavior as expected

I think the toggle would work best (I see no need to hide during print, disabling is good enough!)

sprior, 2022-07-17:
I'll use the vote I don't have to pick the first option.

PR #775 (2022-07-18): feat: Save Config And Restart button visibility toggle

[image]

[image]

Fixes #769
closed 2022-07-18 by pedrolamas