fluidd-core/fluidd #773: Add Bed Mesh visualization to dashboard

inOut47, 2022-07-17:
### Is your feature request related to a problem? Please describe

when priniting, i would like to see the current bed mesh visualization in the main dashboard panel.

### Describe the solution you'd like

add an optional bed mesh visualization widget to the dashbaord layout

matmen, 2022-07-20:
Unfortunately we don't really see that on the roadmap for now because it's just echoing what's already on the tuning page, with no additional functionality or info.. I will close this for now, but please let me know if I missed something here.

inOut47, 2022-07-21:
Fluidd is the UI for klipper and moonraker. i think one of the most important thing in the interface is the UX.

not every feature needs to be something new.
somethings just moving something around can benefit a lot of people.

i am running G29 before every print (as others i am sure).

it will be very helpfull to be able to see the bad state on the dashboard before printing so i can adjust it if its off.

i don't really see any sense in limiting what can and cannot be displayed at the dashboard.
if the widget already exists, dont you think i should be able to show it on the dashboard if i want to?

pedrolamas, 2022-07-21:
Thank you @inOut47, that's a fair point!

To be honest, when @matmen and I discussed this I too thought there was no real advantage in having this as the data is static, but you make a good case when you talk about running G29 on a print startup, so there is value for users with a similar setup.

I've re-opened this ticket so we can assess the implementation of it.

iwm911, 2022-07-26:
Amazing, thank you for the update 🙏

PR #797 (2022-07-28): feat: allow showing Bed Mesh on Dashboard
Desktop:

[image]

Mobile:

[image]

Fixes #773
closed 2022-07-28 by pedrolamas