fluidd-core/fluidd #779: Export stats/history or folders from the Joblist to CSV file

moddok, 2022-07-20:
### Is your feature request related to a problem? Please describe

When I print something for friends I now have to add up the needed time and filament by hand to calculate how much the print will cost.

### Describe the solution you'd like

I would like to export the last X prints from the history or Jobs list to a CSV file to easily add up the needed time and filament.

matmen, 2022-07-20:
You can export the job history as JSON from the [Moonraker API](https://moonraker.readthedocs.io/en/latest/web_api/#history-apis). The corresponding URL to access would be something along of `https://fluiddpi/server/history/list`

matmen, 2022-07-20:
Sorry, I just noticed this was an actual feature request for a download button, not just a question on how to export the data.

closed 2026-10-13 by pedrolamas