fluidd-core/fluidd #780: 'Show current layer' option not persisting on Gcode Preview

alexrmay91, 2022-07-20:
### Fluidd Version

v1.19.0-6184c73

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I never uncheck "Show current layer" in the Gcode Preview window, but I find myself having to re-enable it often.

### What did you expect to happen

I expect the options in the settings drop down to persist.

### How to reproduce

A browser refresh causes 'Show current layer' to become unchecked.

I leave a Fluidd window open on my desktop frequently and have to re-enable it sometimes without me refreshing the page. Maybe it refreshes itself occasionally while I'm away from my desk.

pedrolamas, 2022-07-23:
I've been considering if we shouldn't just have this enabled by default... personally, I see more value having it enabled than not having it at all!

We can still implement serialized state on this (which at the moment does not do, any refresh looses previous state), but I do think that having "show current layer" as a default it does make sense!

Thoughts @matmen @BastelKlug ?

matmen, 2022-07-23:
I think having this be enabled by default would be perfectly fine, I personally always have it enabled (gives a nice progress overview too).

PR #790 (2022-07-26): fix: show current layer in Gcode preview by default
Fixes #780
closed 2022-07-26 by pedrolamas