fluidd-core/fluidd #781: Move Emergency Stop Away from Notifications

bugsysiegals, 2022-07-21:
### Is your feature request related to a problem? Please describe

Yes, I had under power throttle warnings, I went to clear them, and as I was clearing each it allowed me to hit emergency stop killing a full plate of prints at 30% 2 hours into my print.  I'm not sure if I'm the first to do this but it would have been nice to not have these 2 next to each other where I could accidentally hit the emergency stop.

### Describe the solution you'd like

Move the emergency stop button or notifications away from the other buttons you may click?

pedrolamas, 2022-07-21:
Layout is not an easy thing, to be honest one can hit the Emergency Stop button by accident quite easily (especially in a touch screen!)

You can however enable a confirmation dialog on the settings, so when you click that button, you will still have to confirm the action!

bugsysiegals, 2022-07-23:
Thanks, I'd looked to see about confirm in settings but must have missed it, I've enabled and problem solved. Thanks!

closed 2022-07-23 by bugsysiegals