fluidd-core/fluidd #783: Neopixel ColorPicker bug?

SEModer, 2022-07-21:
### Fluidd Version

v1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

When you try to select a color from a connected LED strip, the color starts to change between the selected color and a random one.

### What did you expect to happen

The backlight color of my choice will be set.

### How to reproduce

1) configure led strip
[neopixel my_led]
pin: PA8
chain_count: 14
2) Try pick color from outputs..

### Additional information

https://user-images.githubusercontent.com/789171/180299645-26bbe6e5-5dbd-4c43-9cd2-f60681ee6000.mp4

matmen, 2022-07-21:
Duplicate of #765, will be fixed in the next release by #766

invidtiv, 2022-07-26:
meanwwhile  if you click on ui at the same color for several quick clicks, it stops the behaviour .

closed 2022-08-02 by pedrolamas