fluidd-core/fluidd #788: tracking: camera namespace migration

matmen, 2022-07-25:
For tracking:
Currently, our webcam settings are stored in the `fluidd` namespace under the `cameras` key. This breaks part of the timelapse settings (camera selection) because the moonraker-timelapse plugin cannot find the specified camera.
We should migrate our internal fluidd cameras object to be compatible with [Moonraker's specification](https://moonraker.readthedocs.io/en/latest/web_api/#webcam-apis); as this will fix multiple issues (the above mentioned bug (https://github.com/mainsail-crew/moonraker-timelapse/issues/52) as well as conformity).

pedrolamas, 2022-08-18:
I've taken a look at implementing this, but unless I'm missing something, seems to me that [Moonraker Timelapse](https://github.com/mainsail-crew/moonraker-timelapse) doesn't yet use the Moonraker `webcam` settings.

matmen, 2022-08-18:
It doesn't use the endpoints, but it does [read from the webcams namespace](https://github.com/mainsail-crew/moonraker-timelapse/blob/main/component/timelapse.py#L57)

pedrolamas, 2022-08-18:
But do the endpoints read from that namespace?

matmen, 2022-08-18:
Looks like it (using [this POST request](https://moonraker.readthedocs.io/en/latest/web_api/#add-or-update-a-webcam)):
[image]

pedrolamas, 2022-08-19:
I've been doing some exploration on this and I've already found some interesting bits:

- As @matmen said, the [Moonraker Webcam APIs](https://moonraker.readthedocs.io/en/latest/web_api/#webcam-apis) uses the "webcams" namespace as storage
- Using the Moonraker API means we are limited on what we can store (model is fixed, we can send more fields, but they will be ignored/excluded)
- The Moonraker API does not provide an "id" for each entity and is instead based on the "name" attribute.
- We can store whatever we want in the "webcams" namespace, but ideally, we can (and should) use the expected model that matches the API, and then add whatever extra attributes we need.
- Actions performed directly to the "webcams" namespace are not immediately reflected on the API, as it stores a cached state of it (and there is no way to force a refresh)
- The "activeCamera" property we have will need to go somewhere else (the "config" state maybe?)

I have a working PoC for migrating from our "fluid/cameras" to "webcams"... but the above is quite tricky!

matmen, 2022-08-20:
Using the webcams namespace directly is perfectly fine with me, and I agree with you that we should stick to the model and add additional mandatory info to it. I'm also fine with the activeCamera prop being moved.
The caching of the namespace is a bummer, but I think that would be an upstream issue. After all it's not a reserved namespace

pedrolamas, 2022-08-20:
Perfect, that matches pretty much what I did on my PoC, so will clean it up and bit and submit a PR later today!

PR #839 (2022-08-22): chore: migrates "fluidd/cameras" to "webcams"
So this completely migrates our data under "fluidd/cameras" to "webcams" - **THIS IS A ONE WAY OPERATION!**

We are not using the new Moonraker Webcams API as this does not allow for extra attributes, so we are accessing the "webcams" namespace directly.

We are however using the new Webcam API to get cameras configured in the Moonraker API (we show these as read-only cameras), and to refresh our store if a camera is added/changes/removed via API.

Fixes #788
closed 2022-08-22 by pedrolamas