fluidd-core/fluidd #791: Temperature chart causing unnecessarily high CPU usage (even on i7-7700k)

scottmudge, 2022-07-26:
### Fluidd Version

v1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

## System:

* Intel I7-7700k @ 4.5 GHz 
* 32 GB of DDR4, @ 3200 MHz 
* Intel HD 630 iGPU
* Linux (DietPi with XFCE)

## Software
* Fluidd v1.19.0
* Chromium (latest) -**OR**- Firefox

When the temperature chart is visible, it is causing unnecessarily high CPU usage... roughly ~15-18% total CPU capacity when visible, vs 0-1% usage when not visible.

I've configured chromium to use hardware acceleration (also tried software acceleration), and have tried Firefox with similar tweaks, and both browsers exhibit the same behavior.

I've seen one other (issue #142) from over a year ago, but it was noted that this issue would be addressed in the next release way back then.

I leave the chart hidden for now, but is there an alternative chart API or perhaps a configuration for the chart which can be modified to reduce the CPU usage?

I'm not sure if it has to do with the chart rendering, or perhaps translation/conversion of the chart data being updated, but it's unfortunate. 

I *also* see this behavior on my separate Windows machine (8% usage vs 0%) with Chromium and a i7-8800k clocked at 5.1 GHz. 

### What did you expect to happen

Minor or imperceptible increase in CPU usage, similar to OctoPrint.

### How to reproduce

Just turn the chart on.

pedrolamas, 2022-07-26:
Maybe I'm missing something here, but I just checked this in Windows (both Edge and Chrome) and CPU is running just fine (well below 7% usage from the quick I monitored it), could this be a Linux driver issue?

pedrolamas, 2022-07-26:
> I _also_ see this behavior on my separate Windows machine (8% usage vs 0%) with Chromium and a i7-8800k clocked at 5.1 GHz.

Ah, just read this paragraph... I think this is a bit expected from the eCharts component, not much one can do about it that wouldn't involve replacing with something else that could be even worse!

scottmudge, 2022-07-26:
> Maybe I'm missing something here, but I just checked this in Windows (both Edge and Chrome) and CPU is running just fine (well below 7% usage from the quick I monitored it), could this be a Linux driver issue?

Do you see a difference hiding vs. showing the chart? 8% total CPU resources is still a lot for a single chart.

scottmudge, 2022-07-26:
Eh, looks like a common issue for JS charting software. Closed source "LightningCharts" seems to highlight this problem in an attempt to advertise their product.

Looks like ECharts is already the highest version (5.3.3), so I guess not much can really be done beyond reducing the poll/update rate a bit.

pedrolamas, 2022-07-26:
> Looks like ECharts is already the highest version (5.3.3), so I guess not much can really be done beyond reducing the poll/update rate a bit.

We are already only updating once per second (I believe that was actually the fix on the other issue!), don't think increasing that will help much...

This is a run from my own machine:

[image]

The small red bit I highlighted is when the chart was off, so yes, that takes CPU down to 0, but when it was on, on that capture it never went above 6%!

We can look into alternatives, but given the above I have to admit that is not a priority right now...

scottmudge, 2022-07-26:
Maybe Edge has the *edge* here. Wonder if the build of chromium I'm using has more overhead on the rasterization/rendering end, particularly on Linux. 

Alright, well appreciate you looking into it. I'll close the issue for now and just leave the chart off until I need it.

closed 2022-07-26 by scottmudge