fluidd-core/fluidd #803: Create a probe Z offset wizard

iwm911, 2022-07-29:
### Is your feature request related to a problem? Please describe

Hi all,

From time to time it is required to calibrate the probe Z offset.
Mainly after changing nozzle.

It's possible today to calibrate the probe Z offset using the Probe_calibrate command and then start with the TESTZ z=-1 etc.

For me it is not a problem, however for my 70 year old father it's a bit more complicated and error prone.

He understands the whole paper between the nozzle and the bed process and was able to execute it while he was using marlin without a probe by using the bed screws, however once commands are introduced he is some how get out of balance. 

### Describe the solution you'd like

It could be very helpful for my dad and a lot of newbies (and probably for me to) to be able to perform the Probe_calibrate process using the GUI instead of the CLI.

It can something pretty basic and straightforward which can give the ability to start the calibration process and adjust the Z offset up and down (basically perform the TESTZ command) using simple buttons instead typing commands. 

### Additional information

I would be more than happy to help coding or any thing else, I have some experience with react.

pedrolamas, 2022-07-29:
You're in luck! We've just added a Manual Probe helper dialog that will help with this!

https://github.com/fluidd-core/fluidd/pull/785

It will be available on Fluidd 1.19.1, to be release soon!!

iwm911, 2022-07-29:
So cool!
My dad will really enjoy to be able to do it on his own!.

Thank you guys for all your hard work!

closed 2022-07-29 by iwm911