fluidd-core/fluidd #804: Upload & Start button

BastelKlug, 2022-07-29:
### Is your feature request related to a problem? Please describe

Uploading and printing .gcode files takes too long.

### Describe the solution you'd like

Add a button like on Duet Web Control to start prints faster.
I think it will the best in a similar style to the notification bell

Like this:

[image]

or:

[image]

### Additional information

[image]

BastelKlug, 2022-07-30:
I personally like figure A more, but it looks strange with the save & restart button.

In the worst case it looks like this:
[image]

Any suggestions? @matmen @pedrolamas

pedrolamas, 2022-07-30:
Not fond of having it in blue, that's my main take!

Having said that, I think that that area is starting to get a bit overcrowded, so we need to start considering that too.

PR #974 (2022-12-14): feat: adds Upload and Print button to AppBar
Adds a new optional Upload and Print button to the top navigation bar.

[image]

[image]

[image]

Button is hidden while printing something.

This button visibility is configurable in the settings, and defaults to visible:

[image]

Closes #804
closed 2022-12-14 by pedrolamas