fluidd-core/fluidd #805: Fluidd not always updating information with new print

fredconex, 2022-07-29:
### Fluidd Version

v1.19.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

When using SuperSlicer to send a new file to print (not tested with other slicers) it does send the new file but some info on fluidd does not update with new file/print.

### What did you expect to happen

Loading a new print should update info, times and thubnail.

### How to reproduce

Can't say for sure, but I got it happening randomly when just changing objects on the plate and sending to print, filename remain the same and just objects/gcode change, maybe as filename is same its not understanding that it's a new print?

The print is under progress, fluidd didn't display correct information
[image]

So I've pressed refresh button on browser, then it updated the information
[image]

sarthurdev, 2022-08-11:
Noticed the same issue when using gcode preview. If restarting a print with the same filename but having changed settings in slicer, the gcode viewer won't reload and load button remains disabled.

pedrolamas, 2022-08-15:
I've found and fixed the issue reported by @fredconex (there's now a PR going under review with the fix).

PR #830 (2022-08-17): fix: mitigates undefined history errors
Fixes #805 

The main issue here was the expression `!current_file.history && !current_file.history.filament_used` would throw when `current_file.history` was `null` or `undefined`...

I refactored the rest of the checks a bit to make them clearer.
closed 2022-08-17 by pedrolamas