fluidd-core/fluidd #806: Power toggle in top navigation missing options

ReinBentdal, 2022-07-30:
### Fluidd Version

1.19.0

### Browser

Safari

### Device

Laptop

### Operating System

macOS

### What happened

There is no option to turn on the navigation menu power toggle, the only option is `none`. Am I missing something or is it a bug? Just updated from 1.18.2.
[image]

### What did you expect to happen

Expected there to be an option to turn it on, maybe a new config option which is missing due to upgrading?

### How to reproduce

Might encounter the issue if upgrading from a previous fluid version

BastelKlug, 2022-07-30:
Hi, do you see an element like this?
[image]

ReinBentdal, 2022-07-31:
I am currently in a house-moving process, might be able to check in a couple of days

ReinBentdal, 2022-08-06:
[image]
i dont

matmen, 2022-08-06:
Can you share your power device configuration?

ReinBentdal, 2022-08-07:
not sure if I understand which files you mean. Could you elaborate?

matmen, 2022-08-07:
Both your printer.cfg as well as moonraker.conf

ReinBentdal, 2022-08-07:
**printer.cfg**
```cfg
# This file contains common pin mappings for the BigTreeTech Octopus V1.
# To use this config, the firmware should be compiled for the STM32F446 with a "32KiB bootloader"
# Enable "extra low-level configuration options" and select the "12MHz crystal" as clock reference

# after running "make", copy the generated "klipper/out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the OctoPus with that SD card.

# See docs/Config_Reference.md for a description of parameters.

## Voron Design VORON2 250/300/350mm BigTreeTech OctoPus V1 TMC2209 UART config

[include macros.cfg]
[include fluidd.cfg]

[include stealthburner.cfg]

[include klicky/klicky-probe.cfg]

[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None

[resonance_tester]
accel_chip: adxl345
probe_points:
   125, 125, 20  # an example

#[input_shaper]
#shaper_freq_x: 58.6
#shaper_type_x: mzv
#shaper_freq_y: 45.0
#shaper_type_y: mzv

[mcu]
##  Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
##--------------------------------------------------------------------
serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_130047000350534E4E313020-if00
restart_method: command
##--------------------------------------------------------------------

[printer]
kinematics: corexy
max_velocity: 800
max_accel: 6000            
max_accel_to_decel: 6000
max_z_velocity: 20          #Max 15 for 12V TMC Drivers, can increase for 24V
max_z_accel: 350
square_corner_velocity: 5.0

#####################################################################
#   X/Y Stepper Settings
#####################################################################

##  B Stepper - Left
##  Connected to MOTOR_0
##  Endstop connected to DIAG_0
[stepper_x]
step_pin: PF13
dir_pin: PF12
enable_pin: !PF14
rotation_distance: 40
microsteps: 128
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin: !PG6
position_min: 0
##--------------------------------------------------------------------
[... 527 lines cut ...]

[file_manager]
config_path: ~/klipper_config
log_path: ~/klipper_logs

[data_store]
temperature_store_size: 600
gcode_store_size: 1000

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
enable_auto_refresh: True

# this enabled fluidd updates
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd
```

matmen, 2022-08-07:
Well, what device are you expecting to show up? I don't see any configured in your configs

ReinBentdal, 2022-08-07:
Then it might be on me. I was looking for a way to turn on the button added in #736. Do I then have to add something manually to the configs?

matmen, 2022-08-07:
This is intended as a power toggle for the printer (or any other moonraker power device, really). You will need to add a [power device] to your moonraker config for it to show up

ReinBentdal, 2022-08-07:
i see, then it should be all good. Thanks!

closed 2022-08-07 by ReinBentdal