fluidd-core/fluidd #807: Fluidd unable to save config files opened via link

XoBoR, 2022-07-31:
### Fluidd Version

 v1.19.0-6184c73

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I'm running multiple printers off the same host, so I have a shared_macros.cfg file that I include in all my printers config files.

In order to be able to open this file with the Fluidd editor, I have created links to this file in all the printers config directories, which works. 

However, when I edit the file and try to save it, I get "BAD REQUEST". 

It would be great, and I assume not hard to implement, if this was possible!

Greetings & Thanks.

### What did you expect to happen

Saving the changes to the file linked to

### How to reproduce

Create a config file, then create a link to that file in the printer config directory, edit it via the Fluidd editor and try to hit "Save"

pedrolamas, 2022-08-01:
Thanks for reporting this @XoBoR.

I've managed to reproduce and confirm this bug, however Fluidd is just a web interface that runs on the browser.

This type of issues needs to be reported directly on the [Moonraker repo](https://github.com/Arksine/moonraker/issues), the API Server Fluidd (and Mainsail) use.

pedrolamas, 2022-08-01:
A small addendum to this, I've found the bit of code in Moonraker causing this issue:

https://github.com/Arksine/moonraker/blob/366e68c8f4d83f00a270864ed5064d717db3105d/moonraker/components/file_manager/file_manager.py#L576-L577

The error message above matches what I found on my "moonraker.log" when I was testing:

```
2022-08-01 08:35:30,386 [web.py:log_exception()] - 400 POST /server/files/upload (172.17.0.1): Cannot overwrite symlink: /printer/klipper_config/printer.cfg
2022-08-01 08:35:30,388 [app.py:log_request()] - 400 POST /server/files/upload (172.17.0.1) [_TRUSTED_USER_] 47.61ms
```

closed 2022-08-01 by pedrolamas