fluidd-core/fluidd #810: Add 'Full Update' to Software Updates, that installs all pending updates in the correct order

ChaosBlades, 2022-08-03:
### Is your feature request related to a problem? Please describe

There is no way to apply all updates in Fluidd you need to install them one by one and according to some comments I remember seeing there could be issues if you update in the wrong order where one thing depends on an update from something else which can cause breaking changes.

### Describe the solution you'd like

KlipperScreen has a 'Full Update' button that applies all pending updates in the correct order. I would like to see something similar in Fluidd.

### Additional information

https://github.com/jordanruthe/KlipperScreen/blob/master/panels/system.py

pedrolamas, 2022-08-03:
Thank you for posting this @ChaosBlades

I've taken a look at the file you shared (thank you for providing that), and from what I can see, all it does is update the components **in alphabetical order** (as seen [here](https://github.com/jordanruthe/KlipperScreen/blob/835803d329e093594f4f9e66fb42ec627fad608a/panels/system.py#L58)), so no special "correct order" at all!

As far as I am aware, there is no "correct order" in terms of updating or installing Klipper, at least not one that is on the documentation.

ChaosBlades, 2022-08-03:
Actually I think it might not be needed if the script is doing it all in the background. I think the comments I read about it breaking is if you use it on the WebUI and you install the wrong update you could be left in a broken state and unable to click the button to apply the next update but if it is handled by the script in the background then that would not be the case.

Regardless it would be nice to have a one button solution.

pedrolamas, 2022-08-03:
An "update all" button might be possible but does bring some challenges in terms of updating Moonraker... guess that we would need for it to be the last one to update, as we can't do anything without it running!

BastelKlug, 2022-08-08:
Hi, moonraker has the function to perform a full update: https://moonraker.readthedocs.io/en/latest/web_api/#perform-a-full-update.

PR #817 (2022-08-10): feat(ui): update all feature
Closes: #810 
Demo:

[image]
closed 2022-08-10 by pedrolamas