fluidd-core/fluidd #821: Blank screen when opening file editor

cobyn, 2022-08-09:
### Fluidd Version

1.19.1

### Browser

Edge

### Device

Desktop PC

### Operating System

Windows

### What happened

Openeding a file in configuration shows a blank screen, looking at console log i see this error every time i open a file.
[image]
 
This only is happening on my windows 11 machine running Chromium Edge seem to work fine on android chrome and windows 10 edge. 

### What did you expect to happen

file editor opens with the content of the file

### How to reproduce

try opening a file in configuration manager while on edge on windows 11

pedrolamas, 2022-08-09:
Thank you for reporting this, I have an idea of what this might be!

Please check the following:

- In the address bar, click the small icon that appears just left of the address
- On the popup, click the "Enhance security for this site"
- Disable the "Use enhanced security for this site"
- Refresh the page and try again.

Can you follow these steps and report back if that fixed the issue please?

cobyn, 2022-08-10:
Nice that fixed it!

pedrolamas, 2022-08-10:
Thanks for confirming it, I had suspected this was the case as Microsoft seems to have changed what non-HTTPS sites can do (and in this case, block them from running Web Assembly bits)

Guess we need to add some new instructions for Edge browser...

closed 2022-08-13 by pedrolamas