fluidd-core/fluidd #834: Filter "ok" messages from the terminal

lettore, 2022-08-15:
### Is your feature request related to a problem? Please describe

I use on the same printer Fluidd and Octoprint as klipper Web-ui. When I print using Octoprint in Fluidd I see hundreds of "ok" responses every minute from Klipper, this makes the terminal unusable and also slow down the browser as after some hours it get stuck. Instead when I use Fluidd as print manager often when the print finishes I get "Not SD Printing" message every second.
Parsing all these messages obviously causes a lot of load on the PC.

### Describe the solution you'd like

A check box like the one already present to hide the temperature messages for hide ok and similar messages.

pedrolamas, 2022-08-15:
You can easily add custom filters to Fluidd console view: go to Settings, select Console, and click Add Filter.

You will see a dialog like this:

[image]

Then just fill that and click Save. The filter will then show up on the Console, already selected.

lettore, 2022-08-17:
Yes it's possible and it works.

closed 2022-08-17 by lettore